zayfod/pycozmo #1: Syntax error when launching any test files

CozmoCyke, 2019-08-25:
Hello,

I have Cozmo with last update. Windows 10 , PyCozmo 0.2.0 with Python 3.5.4
I have installed Python on its charging platform. 
I have rised and lowered his lift for discovering its key. 
I'm connecting on Cozmo WIFI point via PC.

Steps :  
1 Open commandline console
2. >pip install pycozmo
3. >python display.py

C:\dev\cozmo\pycozmo\examples>python display.py
Traceback (most recent call last):
  File "display.py", line 5, in <module>
    import pycozmo
  File "C:\Users\didie\AppData\Local\Programs\Python\Python35\lib\site-packages\pycozmo\__init__.py", line 3, in <module>
    from .client import ROBOT_ADDR, Client
  File "C:\Users\didie\AppData\Local\Programs\Python\Python35\lib\site-packages\pycozmo\client.py", line 105
    self.sock: socket.socket = sock
             ^
SyntaxError: invalid syntax

zayfod, 2019-08-25:
OK, you are right. This is syntax that is not suppoted in Python 3.5 and I have been developing Python with 3.7.

Lucklily, this is not really necessary, so I fixed it with ef5df564

I'll set up automated tests with different Python versions soon.

Thanks!

closed 2019-08-25 by zayfod

zayfod, 2019-09-04:
Added automated tests with e9928fc .