zayfod/pycozmo #2: Second test gives Cozmo 01 on Cozmo's screen

CozmoCyke, 2019-08-25:
I've installed new master version.

Steps : 
1. Launch display.py file onetime
C:\dev\cozmo\pycozmo\examples>python display.py
Sending hello...
Connected!
Got  Connect()
Got  UnknownCommand(c9, 33:44:30:09:05:00)
Got  UnknownCommand(ee, 33:44:bd:01:7b:22:76:65:72:73:69:6f:6e:22:3a:20:32:33:38:31:2c:20:22:67:69:74:2d:72:65:76:22:3a:20:22:34:30:38:64:32:38:61:37:66:36:65:36:38:63:62:62:35:62:32:39:63:31:64:63:64:38:63:38:64:62:32:62:33:38:66:39:63:38:63:65:22:2c:20:22:64:61:74:65:22:3a:20:22:54:75:65:20:4a:61:6e:20:20:38:20:31:30:3a:32:37:3a:30:35:20:32:30:31:39:22:2c:20:22:74:69:6d:65:22:3a:20:31:35:34:36:39:37:32:30:32:35:2c:20:22:6d:65:73:73:61:67:65:45:6e:67:69:6e:65:54:6f:52:6f:62:6f:74:48:61:73:68:22:3a:20:22:39:65:34:61:39:36:35:61:63:65:34:65:30:39:64:38:36:39:39:37:62:38:37:62:61:31:34:32:33:35:64:35:22:2c:20:22:6d:65:73:73:61:67:65:52:6f:62:6f:74:54:6f:45:6e:67:69:6e:65:48:61:73:68:22:3a:20:22:61:32:35:39:32:34:37:66:31:36:32:33:31:64:62:34:34:30:39:35:37:32:31:35:62:61:62:61:31:32:61:62:22:2c:20:22:62:75:69:6c:64:22:3a:20:22:44:45:56:45:4c:4f:50:4d:45:4e:54:22:2c:20:22:77:69:66:69:53:69:67:22:3a:20:22:36:39:63:61:30:33:33:35:32:65:34:32:31:34:33:64:33:34:30:66:30:66:37:66:61:63:30:32:65:64:38:66:66:39:36:65:66:31:30:62:22:2c:20:22:72:74:69:70:53:69:67:22:3a:20:22:33:36:35:37:34:39:38:36:64:37:36:31:34:34:61:37:30:65:39:32:35:32:61:62:36:33:33:62:65:34:36:31:37:61:34:62:63:36:36:31:22:2c:20:22:62:6f:64:79:53:69:67:22:3a:20:22:36:39:35:62:35:39:65:66:66:34:33:36:36:34:61:63:64:31:61:35:61:39:35:36:64:30:38:63:36:38:32:62:33:66:38:62:64:32:63:38:22:7d)
Got  UnknownCommand(b0, 01:00:00:00:01:00:02:01:04:00:00:00)
Got  UnknownCommand(b0, 72:02:00:00:9b:01:02:01:04:00:00:00)
Got  UnknownCommand(b0, 5b:01:00:00:9d:01:02:01:01:00:00:00)
Got  UnknownCommand(b0, 78:02:00:00:b3:04:03:01:05:00:00:00)
Got  UnknownCommand(ec, ef:60:15:00)
Got  UnknownCommand(b0, 31:01:00:00:e4:00:02:00)
Sending enable...
Sent UnknownCommand(25, )
Sent UnknownCommand(4b, c4:b6:39:00:00:00:a0:c1)
Sent UnknownCommand(9f, )
Got  UnknownCommand(b0, 31:01:00:00:64:00:01:00)
Got  UnknownCommand(b0, 5b:01:00:00:9e:01:02:01:00:00:00:00)
Got  RobotDelocalized()
Got  UnknownCommand(b0, 43:02:00:00:8a:01:02:03:b2:9f:32:bf:63:7f:92:3e:c2:a7:8b:bf)
Got  UnknownCommand(d1, 02:01:00)
Got  UnknownCommand(d1, 03:01:00)
Got  UnknownCommand(d1, 02:00:00)
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[63, 63])
Sent NextFrame()
Sent DisplayImage(image=[17, 156, 182, 164, 152, 190, 64, 148, 198, 64, 144, 206, 91, 148, 198, 156, 152, 190, 160, 156, 182, 6, 160, 174, 64, 156, 182, 64, 152, 190, 93, 156, 182, 64, 160, 174, 27])
Got  UnknownCommand(b0, 31:01:00:00:1a:01:01:00)
Got  UnknownCommand(d1, 03:00:00)
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[22, 160, 182, 65, 156, 190, 64, 152, 198, 91, 156, 190, 156, 160, 182, 64, 6, 164, 174, 164, 160, 182, 64, 156, 190, 64, 152, 198, 91, 156, 190, 64, 160, 182, 64, 22])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[17, 156, 182, 164, 152, 190, 64, 148, 198, 64, 144, 206, 91, 148, 198, 156, 152, 190, 160, 156, 182, 6, 160, 174, 64, 156, 182, 64, 152, 190, 93, 156, 182, 64, 160, 174, 27])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[22, 160, 182, 65, 156, 190, 64, 152, 198, 91, 156, 190, 156, 160, 182, 64, 6, 164, 174, 164, 160, 182, 64, 156, 190, 64, 152, 198, 91, 156, 190, 64, 160, 182, 64, 22])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[17, 156, 182, 164, 152, 190, 64, 148, 198, 64, 144, 206, 91, 148, 198, 156, 152, 190, 160, 156, 182, 6, 160, 174, 64, 156, 182, 64, 152, 190, 93, 156, 182, 64, 160, 174, 27])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[22, 160, 182, 65, 156, 190, 64, 152, 198, 91, 156, 190, 156, 160, 182, 64, 6, 164, 174, 164, 160, 182, 64, 156, 190, 64, 152, 198, 91, 156, 190, 64, 160, 182, 64, 22])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[17, 156, 182, 164, 152, 190, 64, 148, 198, 64, 144, 206, 91, 148, 198, 156, 152, 190, 160, 156, 182, 6, 160, 174, 64, 156, 182, 64, 152, 190, 93, 156, 182, 64, 160, 174, 27])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[22, 160, 182, 65, 156, 190, 64, 152, 198, 91, 156, 190, 156, 160, 182, 64, 6, 164, 174, 164, 160, 182, 64, 156, 190, 64, 152, 198, 91, 156, 190, 64, 160, 182, 64, 22])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[17, 156, 182, 164, 152, 190, 64, 148, 198, 64, 144, 206, 91, 148, 198, 156, 152, 190, 160, 156, 182, 6, 160, 174, 64, 156, 182, 64, 152, 190, 93, 156, 182, 64, 160, 174, 27])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
[... 105 lines cut ...]
Sent DisplayImage(image=[17, 156, 182, 164, 152, 190, 64, 148, 198, 64, 144, 206, 91, 148, 198, 156, 152, 190, 160, 156, 182, 6, 160, 174, 64, 156, 182, 64, 152, 190, 93, 156, 182, 64, 160, 174, 27])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Sent NextFrame()
Sent DisplayImage(image=[22, 160, 182, 65, 156, 190, 64, 152, 198, 91, 156, 190, 156, 160, 182, 64, 6, 164, 174, 164, 160, 182, 64, 156, 190, 64, 152, 198, 91, 156, 190, 64, 160, 182, 64, 22])
Got  UnknownCommand(b0, 31:01:00:00:a9:00:03:00)
Traceback (most recent call last):
  File "display.py", line 39, in <module>
    main()
  File "display.py", line 32, in main
    time.sleep(1)
KeyboardInterrupt

2. Launch display.py
C:\dev\cozmo\pycozmo\examples>python display.py
Sending hello...
Traceback (most recent call last):
  File "display.py", line 39, in <module>
    main()
  File "display.py", line 14, in main
    time.sleep(0.2)
KeyboardInterrupt

I have to press ctrl-C because no more lines displayed after Sending hello ...

And Cozmo 001 is displaying on Cozmo's screen ...

zayfod, 2019-08-25:
This is expected at the moment as retry/resend logic is still missing in the client.

If you stop an application with CTRL+C, this terminates them immediately and the client does not close the connection to Cozmo gracefully. An immediate subsequent application start will fail to connect as Cozmo is still trying to communicate with the previous instance of the application.

In such situations, you need to wait for about 5 seconds to get the "Cozmo 01" messege and then start the next application. "Cozmo 01" means that Cozmo has stopped receiving ping messages from a client and it has given up retrying and disconnected on its end.

Retry logic will be added but at the moment I want to get more of the Cozmo features working first.

zayfod, 2019-08-25:
On a separate note, on my Cozmo the message is "COZMO 01" and not "Cozmo 001". Would you be able to send a picture of your Cozmo's display when you get it in such a situation?

CozmoCyke, 2019-08-27:
Sorry. But you're right it's "COZMO 01" :-)

brean, 2019-09-17:
I think it would be best practice to catch the keyboard interrupt and do a proper shut down.

zayfod, 2019-09-17:
@brean agreed. This is already done in all examples by using `pycozmo.run_program()`.

brean, 2019-09-19:
I think you can close this issue then?

zayfod, 2019-09-21:
I was using it to motivate myself to complete packet resending. I agree it is of less priority now so I'll close it.

closed 2019-09-21 by zayfod