zayfod/pycozmo #4: Testing backpack_lights.py works but display many unkown commands

CozmoCyke, 2019-08-25:
C:\dev\cozmo\pycozmo\examples>python backpack_lights.py
Sending hello...
Connected!
Got  Connect()
Got  UnknownCommand(c9, 33:44:30:09:05:00)
Got  UnknownCommand(ee, 33:44:bd:01:7b:22:76:65:72:73:69:6f:6e:22:3a:20:32:33:38:31:2c:20:22:67:69:74:2d:72:65:76:22:3a:20:22:34:30:38:64:32:38:61:37:66:36:65:36:38:63:62:62:35:62:32:39:63:31:64:63:64:38:63:38:64:62:32:62:33:38:66:39:63:38:63:65:22:2c:20:22:64:61:74:65:22:3a:20:22:54:75:65:20:4a:61:6e:20:20:38:20:31:30:3a:32:37:3a:30:35:20:32:30:31:39:22:2c:20:22:74:69:6d:65:22:3a:20:31:35:34:36:39:37:32:30:32:35:2c:20:22:6d:65:73:73:61:67:65:45:6e:67:69:6e:65:54:6f:52:6f:62:6f:74:48:61:73:68:22:3a:20:22:39:65:34:61:39:36:35:61:63:65:34:65:30:39:64:38:36:39:39:37:62:38:37:62:61:31:34:32:33:35:64:35:22:2c:20:22:6d:65:73:73:61:67:65:52:6f:62:6f:74:54:6f:45:6e:67:69:6e:65:48:61:73:68:22:3a:20:22:61:32:35:39:32:34:37:66:31:36:32:33:31:64:62:34:34:30:39:35:37:32:31:35:62:61:62:61:31:32:61:62:22:2c:20:22:62:75:69:6c:64:22:3a:20:22:44:45:56:45:4c:4f:50:4d:45:4e:54:22:2c:20:22:77:69:66:69:53:69:67:22:3a:20:22:36:39:63:61:30:33:33:35:32:65:34:32:31:34:33:64:33:34:30:66:30:66:37:66:61:63:30:32:65:64:38:66:66:39:36:65:66:31:30:62:22:2c:20:22:72:74:69:70:53:69:67:22:3a:20:22:33:36:35:37:34:39:38:36:64:37:36:31:34:34:61:37:30:65:39:32:35:32:61:62:36:33:33:62:65:34:36:31:37:61:34:62:63:36:36:31:22:2c:20:22:62:6f:64:79:53:69:67:22:3a:20:22:36:39:35:62:35:39:65:66:66:34:33:36:36:34:61:63:64:31:61:35:61:39:35:36:64:30:38:63:36:38:32:62:33:66:38:62:64:32:63:38:22:7d)
Got  UnknownCommand(b0, 70:02:00:00:99:01:02:06:2e:00:00:00:3a:00:00:00:e8:00:00:00:03:00:00:00:53:00:00:00:e6:00:00:00)
Got  UnknownCommand(b0, 72:02:00:00:9b:01:02:01:08:00:00:00)
Got  UnknownCommand(b0, 5b:01:00:00:9d:01:02:01:02:00:00:00)
Got  UnknownCommand(b0, 78:02:00:00:b3:04:03:01:05:00:00:00)
Got  UnknownCommand(ec, ef:60:15:00)
Got  UnknownCommand(b0, 31:01:00:00:e4:00:02:00)
Sending enable...
Sent UnknownCommand(25, )
Sent UnknownCommand(4b, c4:b6:39:00:00:00:a0:c1)
Sent UnknownCommand(9f, )
Got  UnknownCommand(b0, 31:01:00:00:64:00:01:00)
Got  UnknownCommand(b0, 5b:01:00:00:9e:01:02:01:00:00:00:00)
Got  UnknownCommand(d1, 02:01:00)
Got  UnknownCommand(d1, 03:01:00)
Got  RobotDelocalized()
Got  UnknownCommand(b0, 43:02:00:00:8a:01:02:03:7c:7b:2d:bf:b7:fa:a3:3e:bd:ae:84:bf)
Got  UnknownCommand(b0, 31:01:00:00:1a:01:01:00)
Got  UnknownCommand(d1, 03:00:00)
Sent LightStateCenter(on_color_top=31744, off_color_top=31744, on_frames_top=0, off_frames_top=0, transmission_on_frames_top=0, transmission_off_frames_top=0, offset_top=0, on_color_middle=31744, off_color_middle=31744, on_frames_middle=0, off_frames_middle=0, transmission_on_frames_middle=0, transmission_off_frames_middle=0, offset_middle=0, on_color_bottom=31744, off_color_bottom=31744, on_frames_bottom=0, off_frames_bottom=0, transmission_on_frames_bottom=0, transmission_off_frames_bottom=0, offset_bottom=0, unknown=0)
Sent LightStateSide(on_color_left=31744, off_color_left=31744, on_frames_left=0, off_frames_left=0, transmission_on_frames_left=0, transmission_off_frames_left=0, offset_left=0, on_color_right=31744, off_color_right=31744, on_frames_right=0, off_frames_right=0, transmission_on_frames_right=0, transmission_off_frames_right=0, offset_right=0, unknown=0)
Got  UnknownCommand(d1, 02:00:00)
Got  UnknownCommand(b0, 31:01:00:00:4a:01:02:00)
Got  RobotPoked()
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:b3:b8:1c:45:e2:a7:7b:c4:25:96:11:46:df:51:d8:3e:df:8c:53:3f:ff:bc:05:3e:00:00:00:00)
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:da:34:aa:43:ef:3d:48:c4:f6:d2:1b:46:d1:14:1e:bf:08:00:17:3f:a5:99:69:3f:00:00:00:00)
Sent LightStateCenter(on_color_top=992, off_color_top=992, on_frames_top=0, off_frames_top=0, transmission_on_frames_top=0, transmission_off_frames_top=0, offset_top=0, on_color_middle=992, off_color_middle=992, on_frames_middle=0, off_frames_middle=0, transmission_on_frames_middle=0, transmission_off_frames_middle=0, offset_middle=0, on_color_bottom=992, off_color_bottom=992, on_frames_bottom=0, off_frames_bottom=0, transmission_on_frames_bottom=0, transmission_off_frames_bottom=0, offset_bottom=0, unknown=0)
Sent LightStateSide(on_color_left=992, off_color_left=992, on_frames_left=0, off_frames_left=0, transmission_on_frames_left=0, transmission_off_frames_left=0, offset_left=0, on_color_right=992, off_color_right=992, on_frames_right=0, off_frames_right=0, transmission_on_frames_right=0, transmission_off_frames_right=0, offset_right=0, unknown=0)
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:6d:48:d9:42:5a:f8:07:45:c8:6d:13:46:65:0c:cc:3d:81:f0:85:bf:d9:69:3d:3e:00:00:00:00)
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:2d:4f:f3:44:52:b2:f4:43:e5:fc:1c:46:02:71:cd:be:0e:bb:2c:be:ee:4e:16:3f:00:00:00:00)
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:be:02:18:45:aa:c7:18:c4:0a:c3:18:46:1c:29:92:be:a1:aa:3c:bf:9f:6c:02:3f:00:00:00:00)
Got  UnknownCommand(b0, 31:01:00:00:4a:01:02:00)
Got  RobotPoked()
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:74:21:40:45:13:68:e8:c3:63:07:14:46:d0:f2:b8:be:05:18:28:3e:6d:58:a9:3f:00:00:00:00)
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:68:c0:2e:44:7c:b6:1f:43:89:e7:16:46:00:b6:45:be:fe:38:31:3f:36:53:6c:3f:00:00:00:00)
Sent LightStateCenter(on_color_top=31, off_color_top=31, on_frames_top=0, off_frames_top=0, transmission_on_frames_top=0, transmission_off_frames_top=0, offset_top=0, on_color_middle=31, off_color_middle=31, on_frames_middle=0, off_frames_middle=0, transmission_on_frames_middle=0, transmission_off_frames_middle=0, offset_middle=0, on_color_bottom=31, off_color_bottom=31, on_frames_bottom=0, off_frames_bottom=0, transmission_on_frames_bottom=0, transmission_off_frames_bottom=0, offset_bottom=0, unknown=0)
Sent LightStateSide(on_color_left=31, off_color_left=31, on_frames_left=0, off_frames_left=0, transmission_on_frames_left=0, transmission_off_frames_left=0, offset_left=0, on_color_right=31, off_color_right=31, on_frames_right=0, off_frames_right=0, transmission_on_frames_right=0, transmission_off_frames_right=0, offset_right=0, unknown=0)
Sent LightStateCenter(on_color_top=32767, off_color_top=32767, on_frames_top=0, off_frames_top=0, transmission_on_frames_top=0, transmission_off_frames_top=0, offset_top=0, on_color_middle=32767, off_color_middle=32767, on_frames_middle=0, off_frames_middle=0, transmission_on_frames_middle=0, transmission_off_frames_middle=0, offset_middle=0, on_color_bottom=32767, off_color_bottom=32767, on_frames_bottom=0, off_frames_bottom=0, transmission_on_frames_bottom=0, transmission_off_frames_bottom=0, offset_bottom=0, unknown=0)
Sent LightStateSide(on_color_left=32767, off_color_left=32767, on_frames_left=0, off_frames_left=0, transmission_on_frames_left=0, transmission_off_frames_left=0, offset_left=0, on_color_right=32767, off_color_right=32767, on_frames_right=0, off_frames_right=0, transmission_on_frames_right=0, transmission_off_frames_right=0, offset_right=0, unknown=0)
Sent LightStateCenter(on_color_top=0, off_color_top=0, on_frames_top=0, off_frames_top=0, transmission_on_frames_top=0, transmission_off_frames_top=0, offset_top=0, on_color_middle=0, off_color_middle=0, on_frames_middle=0, off_frames_middle=0, transmission_on_frames_middle=0, transmission_off_frames_middle=0, offset_middle=0, on_color_bottom=0, off_color_bottom=0, on_frames_bottom=0, off_frames_bottom=0, transmission_on_frames_bottom=0, transmission_off_frames_bottom=0, offset_bottom=0, unknown=0)
Sent LightStateSide(on_color_left=0, off_color_left=0, on_frames_left=0, off_frames_left=0, transmission_on_frames_left=0, transmission_off_frames_left=0, offset_left=0, on_color_right=0, off_color_right=0, on_frames_right=0, off_frames_right=0, transmission_on_frames_right=0, transmission_off_frames_right=0, offset_right=0, unknown=0)
Got  UnknownCommand(b0, 5c:02:00:00:70:01:01:07:ef:5e:bb:43:62:c7:9d:45:b6:1f:01:46:c8:e1:76:3c:98:34:2d:bf:91:18:54:3e:00:00:00:00)
Sent Disconnect()

C:\dev\cozmo\pycozmo\examples>

zayfod, 2019-08-25:
This is expected, but the output does not have to be so verbose...

CozmoCyke, 2019-08-27:
And what mean these unknown commands ? What is the corresponding name or function ?

zayfod, 2019-08-31:
Logging verbosity has been decreased with 5b196a.

closed 2019-08-31 by zayfod