zayfod/pycozmo #5: ImportError: cannot import name 'Type' on dev branch 0.4.0

CozmoCyke, 2019-09-06:
Configuration : 
Python 3.5.1
pycozmo 0.4.0 dev
Windows 7
Cozmo Classic with backpack Button

D:\dev\Cozmo\pycozmo\examples>python backpack_lights.py
Traceback (most recent call last):
  File "backpack_lights.py", line 5, in <module>
    import pycozmo
  File "C:\Python35\lib\site-packages\pycozmo-0.4.0-py3.5.egg\pycozmo\__init__.p
y", line 5, in <module>
    from .run import *
  File "C:\Python35\lib\site-packages\pycozmo-0.4.0-py3.5.egg\pycozmo\run.py", l
ine 8, in <module>
    from . import client
  File "C:\Python35\lib\site-packages\pycozmo-0.4.0-py3.5.egg\pycozmo\client.py"
, line 16, in <module>
    from .frame import Frame
  File "C:\Python35\lib\site-packages\pycozmo-0.4.0-py3.5.egg\pycozmo\frame.py",
 line 4, in <module>
    from .protocol_declaration import FRAME_ID, MIN_FRAME_SIZE, FrameType, Packe
tType
  File "C:\Python35\lib\site-packages\pycozmo-0.4.0-py3.5.egg\pycozmo\protocol_d
eclaration.py", line 7, in <module>
    from .protocol_ast import *
  File "C:\Python35\lib\site-packages\pycozmo-0.4.0-py3.5.egg\pycozmo\protocol_a
st.py", line 9, in <module>
    from typing import List, Optional, Union, Type
ImportError: cannot import name 'Type'

D:\dev\Cozmo\pycozmo\examples>

zayfod, 2019-09-06:
OK, I will update the Python requirement to 3.5.4 - thanks.

Tests with Python 3.5.4 on Windows have passed - https://github.com/zayfod/pycozmo/runs/212767778

`Type` has been added to `typing` in Python 3.5.2.

zayfod, 2019-09-06:
Added requirement for Python 3.5.4 with 5504bb47 .

closed 2019-09-06 by zayfod