zayfod/pycozmo #6: No change for RobotState backpack_touch_sensor_raw

CozmoCyke, 2019-09-07:
I've changed some lines to add backpack button test.
During the test , I'm pressing the backpack button but I didn't see any change in this value :-(

#!/usr/bin/env python

import time

import pycozmo

def on_robot_state(cli, pkt: pycozmo.protocol_encoder.RobotState):
    del cli
    print("Battery level: {:.01f} V".format(pkt.battery_voltage))
    print("Time stamp: {0}".format(pkt.timestamp))

def on_robot_state2(cli, pkt: pycozmo.protocol_encoder.RobotState):
    del cli
    print("backpack touch sensor raw: {0}".format(pkt.backpack_touch_sensor_raw))  

def on_robot_poked(cli, pkt: pycozmo.protocol_encoder.RobotPoked):
    del cli, pkt
    print("Robot poked.")

def on_robot_falling_started(cli, pkt: pycozmo.protocol_encoder.FallingStarted):
    del cli, pkt
    print("Started falling.")

def on_robot_falling_stopped(cli, pkt: pycozmo.protocol_encoder.FallingStopped):
    del cli
    print("Falling stopped after {} ms. Impact intensity {:.01f}.".format(pkt.duration_ms, pkt.impact_intensity))

def on_robot_picked_up(cli, state):
    del cli
    if state:
        print("Picked up.")
    else:
        print("Put down.")

def on_robot_charging(cli, state):
    del cli
    if state:
        print("Started charging.")
    else:
        print("Stopped charging.")

def on_cliff_detected(cli, state):
    del cli
    if state:
        print("Cliff detected.")

def on_robot_wheels_moving(cli, state):
    del cli
    if state:
        print("Started moving.")
    else:
        print("Stopped moving.")

def pycozmo_program(cli):

    cli.add_handler(pycozmo.protocol_encoder.RobotState, on_robot_state, one_shot=True)
    cli.add_handler(pycozmo.protocol_encoder.RobotState, on_robot_state2, one_shot=False)
    cli.add_handler(pycozmo.protocol_encoder.RobotPoked, on_robot_poked)
    cli.add_handler(pycozmo.protocol_encoder.FallingStarted, on_robot_falling_started)
    cli.add_handler(pycozmo.protocol_encoder.FallingStopped, on_robot_falling_stopped)
    cli.add_handler(pycozmo.client.EvtRobotPickedUpChange, on_robot_picked_up)
    cli.add_handler(pycozmo.client.EvtRobotChargingChange, on_robot_charging)
    cli.add_handler(pycozmo.client.EvtCliffDetectedChange, on_cliff_detected)
    cli.add_handler(pycozmo.client.EvtRobotWheelsMovingChange, on_robot_wheels_moving)

    while True:
        try:
[... 271 lines cut ...]
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
backpack touch sensor raw: 0
2019-09-07 07:41:32.834 pycozmo.general      INFO     Done.

D:\dev\Cozmo\pycozmo\examples>

CozmoCyke, 2019-09-07:
I've added the partial log with debug.

backpack touch sensor raw: 0
2019-09-07 08:21:08.148 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=201779560, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.180 pycozmo.protocol     DEBUG    Got  RobotState(timestamp=
3788720, pose_frame_id=0, pose_origin_id=0, pose_x=0.0, pose_y=0.0, pose_z=0.0,
pose_angle_rad=-0.0019488274119794369, pose_pitch_rad=0.04719590023159981, lwhee
l_speed_mmps=0.0, rwheel_speed_mmps=0.0, head_angle_rad=-0.4318157136440277, lif
t_height_mm=-0.19826610386371613, accel_x=-3724.85400390625, accel_y=-47.9003906
25, accel_z=9196.875, gyro_x=-0.0004579275846481323, gyro_y=0.001025638543069362
6, gyro_z=-0.0007229367620311677, battery_voltage=3.81195068359375, status=784,
cliff_data_raw=[873, 0, 0, 0], backpack_touch_sensor_raw=0, curr_path_segment=25
5)
backpack touch sensor raw: 0
2019-09-07 08:21:08.181 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=201809578, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.182 pycozmo.protocol     DEBUG    Got  ObjectAvailable(facto
ry_id=1532612787, object_type=2, rssi=63)
2019-09-07 08:21:08.213 pycozmo.protocol     DEBUG    Got  RobotState(timestamp=
3788750, pose_frame_id=0, pose_origin_id=0, pose_x=0.0, pose_y=0.0, pose_z=0.0,
pose_angle_rad=-0.001948242075741291, pose_pitch_rad=0.047021254897117615, lwhee
l_speed_mmps=0.0, rwheel_speed_mmps=0.0, head_angle_rad=-0.4318157136440277, lif
t_height_mm=-0.19826610386371613, accel_x=-3747.008056640625, accel_y=-56.282958
984375, accel_z=9202.263671875, gyro_x=0.00034121517091989517, gyro_y=0.00075797
1778512001, gyro_z=0.0005833512987010181, battery_voltage=3.81195068359375, stat
us=784, cliff_data_raw=[873, 0, 0, 0], backpack_touch_sensor_raw=0, curr_path_se
gment=255)
backpack touch sensor raw: 0
2019-09-07 08:21:08.214 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=201839587, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.246 pycozmo.protocol     DEBUG    Got  RobotState(timestamp=
3788780, pose_frame_id=0, pose_origin_id=0, pose_x=0.0, pose_y=0.0, pose_z=0.0,
pose_angle_rad=-0.0019508912228047848, pose_pitch_rad=0.04701885208487511, lwhee
l_speed_mmps=0.0, rwheel_speed_mmps=0.0, head_angle_rad=-0.4318157136440277, lif
t_height_mm=-0.19826610386371613, accel_x=-3717.0703125, accel_y=-34.12902832031
25, accel_z=9205.8564453125, gyro_x=0.0006058858707547188, gyro_y=0.000489760190
2484894, gyro_z=-0.0009840140119194984, battery_voltage=3.81195068359375, status
=784, cliff_data_raw=[873, 0, 0, 0], backpack_touch_sensor_raw=0, curr_path_segm
ent=255)
backpack touch sensor raw: 0
2019-09-07 08:21:08.247 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=201869613, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.280 pycozmo.protocol     DEBUG    Got  RobotState(timestamp=
3788810, pose_frame_id=0, pose_origin_id=0, pose_x=0.0, pose_y=0.0, pose_z=0.0,
pose_angle_rad=-0.001960350666195154, pose_pitch_rad=0.04706447198987007, lwheel
_speed_mmps=0.0, rwheel_speed_mmps=0.0, head_angle_rad=-0.4318157136440277, lift
_height_mm=-0.19826610386371613, accel_x=-3708.0888671875, accel_y=-62.869262695
3125, accel_z=9187.8935546875, gyro_x=7.31535255908966e-05, gyro_y=0.00022290088
23633194, gyro_z=0.0011069801403209567, battery_voltage=3.82794189453125, status
=784, cliff_data_raw=[873, 0, 0, 0], backpack_touch_sensor_raw=0, curr_path_segm
ent=255)
backpack touch sensor raw: 0
2019-09-07 08:21:08.282 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=201899632, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.282 pycozmo.protocol     DEBUG    Got  RobotState(timestamp=
3788840, pose_frame_id=0, pose_origin_id=0, pose_x=0.0, pose_y=0.0, pose_z=0.0,
pose_angle_rad=-0.0019324577879160643, pose_pitch_rad=0.04702460393309593, lwhee
l_speed_mmps=0.0, rwheel_speed_mmps=0.0, head_angle_rad=-0.4318157136440277, lif
t_height_mm=-0.19826610386371613, accel_x=-3710.48388671875, accel_y=-35.9252929
6875, accel_z=9188.4921875, gyro_x=-0.000989830121397972, gyro_y=-0.000309749506
4139366, gyro_z=5.9378642617957667e-05, battery_voltage=3.81195068359375, status
=784, cliff_data_raw=[873, 0, 0, 0], backpack_touch_sensor_raw=0, curr_path_segm
ent=255)
backpack touch sensor raw: 0
[... 207 lines cut ...]
784, cliff_data_raw=[873, 0, 0, 0], backpack_touch_sensor_raw=0, curr_path_segme
nt=255)
backpack touch sensor raw: 0
2019-09-07 08:21:08.777 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=202410157, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.807 pycozmo.protocol     DEBUG    Got  RobotState(timestamp=
3789350, pose_frame_id=0, pose_origin_id=0, pose_x=0.0, pose_y=0.0, pose_z=0.0,
pose_angle_rad=-0.0020155729725956917, pose_pitch_rad=0.04712124168872833, lwhee
l_speed_mmps=0.0, rwheel_speed_mmps=0.0, head_angle_rad=-0.4318157136440277, lif
t_height_mm=-0.19826610386371613, accel_x=-3709.28662109375, accel_y=-29.3389892
578125, accel_z=9184.900390625, gyro_x=-0.0007159821689128876, gyro_y=-0.0003175
[... cut ...]
2019-09-07 08:21:08.810 pycozmo.protocol     DEBUG    Got  AnimationState(timest
amp=202440192, num_anim_bytes_played=36, num_audio_frames_played=6, enabled_anim
_tracks=255, tag=0, client_drop_count=0)
2019-09-07 08:21:08.821 pycozmo.general      DEBUG    Disconnecting...
2019-09-07 08:21:08.822 pycozmo.general      DEBUG    Stopping client...
2019-09-07 08:21:08.823 pycozmo.protocol     DEBUG    Sent Disconnect()
2019-09-07 08:21:09.807 pycozmo.general      INFO     Done.

D:\dev\Cozmo\pycozmo\examples>

zayfod, 2019-09-07:
Thanks for trying this.

It seems that `RobotState.backpack_touch_sensor_raw` does not reflect button events. There are no other packets in the second log that may indicate button events also.

Clarified the documentation with a49b1f5d0b.

This is a mystery for the moment...

My two Cozmos don't have buttons, so at the moment I cannot test this.

closed 2019-09-07 by zayfod