zayfod/pycozmo #10: rc.py PyCozmo joystick example doesn't work with Client API

CozmoCyke, 2019-09-22:
i try to use rc.py example in the inputs branch with the last dev branch of PyCozmo.
Before the new Client API , it is working ok. But now with the new Client API I have the following error : 

Windows 7
Game sir G3w with USB cable.

D:\dev\Cozmo\pycozmo\20190921\examples>python rc-inputs.py
Microsoft Keyboard
Microsoft Mouse
Microsoft X-Box 360 pad
2019-09-22 02:22:35.918 root            INFO     Initializing...
2019-09-22 02:22:35.968 pycozmo.general INFO     Firmware version 2381.
2019-09-22 02:22:36.029 pycozmo.general INFO     Body S/N 0x45a0330c.
Traceback (most recent call last):
  File "rc-inputs.py", line 269, in <module>
    main()
  File "rc-inputs.py", line 262, in main
    res = app.init()
  File "rc-inputs.py", line 66, in init
    self.cli.send(pkt)
AttributeError: 'Client' object has no attribute 'send'

zayfod, 2019-09-22:
I think that you have installed an older version with `pip` and you are using the examples from a newer one.

Actually, you are using the `rc-inputs.py` example from the `inputs` branche which was created, before the introduction of the new client API.

Wait a few more days for the final integration of the `inputs` module.

zayfod, 2019-09-22:
The raceback above cannot be reproduced on dev.

I just fixed this with #e58b208c:
```
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/kaloyan/work/pycozmo/examples/rc.py", line 111, in run
    self._handler(event)
  File "/home/kaloyan/work/pycozmo/examples/rc.py", line 318, in _handle_input
    self._drive_wheels(self.speed_left, self.speed_right)
  File "/home/kaloyan/work/pycozmo/examples/rc.py", line 191, in _drive_wheels
    self.cli.drive_wheels(lwheel_speed=lw, rwheel_speed=rw)
  File "/home/kaloyan/work/pycozmo/pycozmo/client.py", line 310, in drive_wheels
    lwheel_accel_mmps2=lwheel_acc, rwheel_accel_mmps2=rwheel_acc)
  File "/home/kaloyan/work/pycozmo/pycozmo/protocol_encoder.py", line 972, in __init__
    self.lwheel_accel_mmps2 = lwheel_accel_mmps2
  File "/home/kaloyan/work/pycozmo/pycozmo/protocol_encoder.py", line 997, in lwheel_accel_mmps2
    self._lwheel_accel_mmps2 = validate_float("lwheel_accel_mmps2", value)
  File "/home/kaloyan/work/pycozmo/pycozmo/protocol_utils.py", line 20, in validate_float
    value = float(value)
TypeError: float() argument must be a string or a number, not 'NoneType'
```

CozmoCyke, 2019-09-22:
I try again with rc-input.py (inputs branch)
rc.py (dev branch) works only on Linux (import evdev) but not on Windows !
I unfortunately the same error :-(

D:\dev\Cozmo\pycozmo\20190922\examples>python rc-inputs.py
Microsoft Keyboard
Microsoft Mouse
Microsoft X-Box 360 pad
2019-09-23 00:39:22.461 root            INFO     Initializing...
2019-09-23 00:39:22.504 pycozmo.general INFO     Firmware version 2381.
2019-09-23 00:39:22.533 pycozmo.general INFO     Body S/N 0x45a0330c.
Traceback (most recent call last):
  File "rc-inputs.py", line 269, in <module>
    main()
  File "rc-inputs.py", line 262, in main
    res = app.init()
  File "rc-inputs.py", line 66, in init
    self.cli.send(pkt)
AttributeError: 'Client' object has no attribute 'send'

CozmoCyke, 2019-09-22:
Hello ,

I found the errors ! :-)
You must use the Client API in the rc.py !

Here's the code who works for me :  

#!/usr/bin/env python

import argparse
import logging
import time
import threading

import inputs

import pycozmo

class InputThread(object):
    """ Thread for reading input. """

    def __init__(self, handler):
        self._stop = False
        self._thread = None
        self._handler = handler

    def start(self):
        self._stop = False
        self._thread = threading.Thread(target=self.run, name="InputThread")
        self._thread.daemon = True
        self._thread.start()

    def stop(self):
        logging.debug("Input thread stopping...")
        self._stop = True
        self._thread.join()
        logging.debug("Input thread joined.")

    def run(self):
        logging.debug("Input thread started.")
        while not self._stop:
            events = inputs.get_gamepad()
            for event in events:
                self._handler(event)
        logging.debug("Input thread stopped.")

class RCApp(object):
    """ Application class. """

    def __init__(self):
        logging.info("Initializing...")
        self._stop = False
        self.input_thread = InputThread(self._handle_input)
        self.cli = pycozmo.Client()
        self.speed = 0.0        # -1.0 - 1.0
        self.steering = 0.0     # -1.0 - 1.0
        self.speed_left = 0.0   # 0 - 1.0
        self.speed_right = 0.0  # 0 - 1.0
        self.lift = True

    def init(self):
        """ Initialize application. """
        self._stop = False
        # Connect to Cozmo
        self.cli.start()
        self.cli.connect()
        self.cli.wait_for_robot()
        # Raise head
        angle = (pycozmo.robot.MAX_HEAD_ANGLE.radians - pycozmo.robot.MIN_HEAD_ANGLE.radians) * 0.1
        #pkt = pycozmo.protocol_encoder.SetHeadAngle(angle_rad=angle)
[... 176 lines cut ...]

def main():
    # Parse command-line.
    args = parse_args()

    # Configure logging.
    level = logging.DEBUG if args.verbose else logging.INFO
    logging.basicConfig(
        format="%(asctime)s.%(msecs)03d %(name)-15s %(levelname)-8s %(message)s",
        datefmt='%Y-%m-%d %H:%M:%S',
        level=level)

    #
    for device in inputs.devices:
        print(device)

    # Create application object.
    app = RCApp()
    res = app.init()
    if res:
        app.run()
        app.term()

if __name__ == '__main__':
    main()

zayfod, 2019-09-23:
OK, I am glad you figured it out.

This is `rc.py` from the inputs branch which is still not integrated in `dev` and now somewhat obsolete.

I'll take your changes when integrating the `inputs` branch some time soon.

Thanks.

closed 2019-09-23 by zayfod