zayfod/pycozmo #13: Unsupported firmware version 10502

paulbaumgarten, 2019-10-31:
I've just started experimenting with the pycozmo library and am attempting to write tutorials for the purposes of using it with my middle school students.

Strangely, code that was working last week has now stopped working with a unsupported firmware version error occurring. The thing is I haven't (consciously) done anything to upgrade/change the firmware on the Cozmo. Have I missed something here? If the firmware has changed, is there a way I can downgrade to the supported version?

```
2019-10-31 15:08:07.360 pycozmo.general      INFO     Firmware version 10502.
2019-10-31 15:08:07.424 pycozmo.general      INFO     Body S/N 0x43225b47.
2019-10-31 15:08:07.429 pycozmo.general      ERROR    Unsupported Cozmo firmware version 10502. Only version 2381 is supported currently.
Traceback (most recent call last):
  File "C:/Users/pab/Desktop/testing.py", line 19, in <module>
    pycozmo.run_program(pycozmo_program)
  File "C:\Python37\lib\site-packages\pycozmo\run.py", line 41, in run_program
    cli.wait_for_robot()
  File "C:\Python37\lib\site-packages\pycozmo\client.py", line 150, in wait_for_robot
    raise exception.UnsupportedFirmwareVersion("Unsupported Cozmo firmware version.")
pycozmo.exception.UnsupportedFirmwareVersion: Unsupported Cozmo firmware version.
```

Thank you in advance.

zayfod, 2019-10-31:
This is strange. All Cozmo firmware versions that I have seen are 4 digit numbers - https://github.com/zayfod/pycozmo/blob/master/docs/versions.md

The Cozmo app automatically updates the Cozmo firmware when a connecton is established. You can also confirm the Cozmo firmware version in the Cozmo app settings menu.

It is strange that you are saying that code worked before. Did you use the same Cozmo unit?

Without knowing more, my best guess is that you are using a Cozmo unit that has not been updated by the Cozmo app and is using an older firmware that makes PyCozmo interpret the FirmwareSignature message incorrectly.

Would you please, run the following minimal code and share the output:
```
import time
import pycozmo

def pycozmo_program(cli: pycozmo.client.Client):
    time.sleep(0.5)

pycozmo.run_program(pycozmo_program, log_level="DEBUG", protocol_log_level="DEBUG")
```

paulbaumgarten, 2019-10-31:
Problem fixed. I re-installed the app on my phone and used it to connect to the Cozmo. As you advised, this automatically updated the firmware on board and has restored it to the correct version. Pycozmo is working again.

I ran the debug prior to connecting with the phone app and include the output here for reference purposes. I infer from the debug output that the Cozmo had been factory reset somehow.

```
2019-11-01 07:39:46.791 pycozmo.general      DEBUG    Starting client...
2019-11-01 07:39:46.795 pycozmo.general      DEBUG    Starting...
2019-11-01 07:39:46.797 pycozmo.general      DEBUG    Connecting...
2019-11-01 07:39:46.799 pycozmo.general      DEBUG    Connecting...
2019-11-01 07:39:46.804 pycozmo.protocol     DEBUG    Got  Connect()
2019-11-01 07:39:46.805 pycozmo.general      DEBUG    Connected.
2019-11-01 07:39:46.838 pycozmo.protocol     DEBUG    Got  HardwareInfo(serial_number_head=92367874, unknown1=5, unknown2=0)
2019-11-01 07:39:46.840 pycozmo.protocol     DEBUG    Got  FirmwareSignature(unknown=27650, signature={"build": "FACTORY", "version": 10502, "date": "Mon Aug  7 09:21:24 2017", "time": 1502122884})
2019-11-01 07:39:46.843 pycozmo.general      INFO     Firmware version 10502.
2019-11-01 07:39:46.845 pycozmo.protocol     DEBUG    Sent Enable()
2019-11-01 07:39:46.847 pycozmo.protocol     DEBUG    Sent Enable()
2019-11-01 07:39:46.850 pycozmo.protocol     DEBUG    Got  DebugData(debug_id=1, unused=0, unknown2=1, unknown3=2, data=[4429])
2019-11-01 07:39:46.852 pycozmo.protocol     DEBUG    Got  DebugData(debug_id=626, unused=0, unknown2=411, unknown3=2, data=[196])
2019-11-01 07:39:46.854 pycozmo.protocol     DEBUG    Got  DebugData(debug_id=347, unused=0, unknown2=413, unknown3=2, data=[1])
2019-11-01 07:39:46.856 pycozmo.protocol     DEBUG    Got  DebugData(debug_id=632, unused=0, unknown2=1203, unknown3=3, data=[5])
2019-11-01 07:39:46.858 pycozmo.protocol     DEBUG    Got  UnknownCommand(ec, ef:60:15:00)
2019-11-01 07:39:46.860 pycozmo.protocol     DEBUG    Got  DebugData(debug_id=305, unused=0, unknown2=228, unknown3=2, data=[])
2019-11-01 07:39:46.871 pycozmo.protocol     DEBUG    Got  BodyInfo(serial_number=1126325063, body_hw_version=5, body_color=BodyColor.LE_BL_v16)
2019-11-01 07:39:46.873 pycozmo.general      INFO     Body S/N 0x43225b47.
2019-11-01 07:39:46.875 pycozmo.general      ERROR    Unsupported Cozmo firmware version 10502. Only version 2381 is supported currently.
Traceback (most recent call last):
  File "C:/Users/pab/Desktop/cozmo-debug.py", line 7, in <module>
    pycozmo.run_program(pycozmo_program, log_level="DEBUG", protocol_log_level="DEBUG")
  File "C:\Python37\lib\site-packages\pycozmo\run.py", line 41, in run_program
    cli.wait_for_robot()
  File "C:\Python37\lib\site-packages\pycozmo\client.py", line 150, in wait_for_robot
    raise exception.UnsupportedFirmwareVersion("Unsupported Cozmo firmware version.")
pycozmo.exception.UnsupportedFirmwareVersion: Unsupported Cozmo firmware version.
```

Thank you

closed 2019-10-31 by paulbaumgarten