zayfod/pycozmo #14: Laser scan

severin-lemaignan, 2019-11-28:
Any idea whether we can access to the laser scans?

The Cozmo SDK exports an occupancy map (I assume built internally from the laser scans), but does not mention anywhere an access to the raw scans (which would be tremendously useful for ROS integration)

severin-lemaignan, 2019-11-28:
[I'm just realising that I'm referring to Vector, not Cozmo here... Does pycozmo supports Vector?]

CozmoCyke, 2019-11-29:
Hello Séverin ,
No. PyCozmo is supporting only the Anki Cosmo Robot.
Perhaps one day we can get a PyVector ;-)

zayfod, 2019-12-01:
Cozmo does not have a laser range detector.

PyCozmo targets only Anki Cozmo, due to the fact that the SDK communication interfaces of Cozmo and Vector differ conceptually and significantly.

closed 2019-12-01 by zayfod