zayfod/pycozmo #16: Self docking on charger

maxic62, 2019-12-03:
Hello,

Thanks for your work, I appreciate a lot to connect to my Cozmo directly without any android device ;) 🥇 

I just discovered this script based on cozmosdk that allow Cozmo to self dock. It seems that this script is using internal image recognition :
[https://forums.anki.com/t/program-self-docking-on-charger/13830](url)

Do you think it is possible to have the same functionnality here with pycozmo ?? It would be nice to can charge it automatically

Thanks
Max

zayfod, 2019-12-05:
You are correct - the script uses at least the following features, implemented in the Cozmo app which are thus not available in PyCozmo today:
- text-to-speech
  - `robot.say_text()`
- image recognition
  - `robot.world.wait_for_observed_light_cube()`
- path planning
  - `robot.place_object_on_ground_here()`
  - `robot.place_on_object()`
  - `robot.pickup_object()`
  - `robot.go_to_object()`
- behaviors
  - `robot.start_behavior()`

From my perspective, such features do not belong to PyCozmo, which provides only "low-level communication" with the robot.

I am planning to start working on a new project in the near future though, code named `PyCozmoBrain`. I think of it as a replacement for the Cozmo app, written in Python and using PyCozmo for communicating with the robot. It will provide at least:
- image recognition - cubes, pets, people, etc.
- object tracking
- path planning

Help is welcome!

maxic62, 2019-12-07:
Hello zayfod,

I agree completly your approach.

I have experience in SW dev and python so It will be a pleasure to help you if you want !
Will opencv be mainly used for pyCozmoBrain or tensorflow maybe ?

See You,
Max

zayfod, 2019-12-08:
Image recognition will definitely be based on Tensorflow. OpenCV may still come handy for image manipulation though.

closed 2019-12-08 by zayfod

HomeAssistantTycoon, 2025-12-24:
How is PyCozmoBrain coming along?