zayfod/pycozmo #18: wait_for_robot() throws connection error despite having connected

synicalsyntax, 2020-01-26:
Code:
```py
import pycozmo
import time

pycozmo.setup_basic_logging(log_level="DEBUG", protocol_log_level="DEBUG")
cli = pycozmo.Client()
cli.start()
cli.connect()
cli.wait_for_robot()
cli.set_head_angle(angle=0.6)
time.sleep(2)
cli.set_head_angle(angle=0.2)
```
Error:
```
Traceback (most recent call last):
  File "test.py", line 8, in <module>
    cli.wait_for_robot() 
  File "C:\path\venv\cozmo\lib\site-packages\pycozmo\client.py", line 157, in wait_for_robot
    raise exception.ConnectionTimeout("Failed to connect to Cozmo.")
pycozmo.exception.ConnectionTimeout: Failed to connect to Cozmo.
```
Debug logs: https://paste.mod.gg/udesumuxef.log 
* Note that client does connect successfully and hardware info, animation state, and image imu data events are all received

Dependencies:
```
Package     Version 
----------- ------- 
cozmoclad   3.4.0
dpkt        1.9.2
flatbuffers 1.11
numpy       1.18.1
Pillow      7.0.0
pip         19.2.3
pycozmo     0.6.0
```
Network info (connected before running code):
```
SSID:	Cozmo_31ABCF
Protocol:	802.11g
Security type:	WPA2-Personal
Network band:	2.4 GHz
Network channel:	3
Link-local IPv6 address:	fe80::1954:c4de:83b3:5abb%4
IPv4 address:	172.31.1.2
IPv4 DNS servers:	172.31.1.1
Manufacturer:	Intel Corporation
Description:	Intel(R) Wireless-AC 9260 160MHz
Driver version:	21.60.0.5
Physical address (MAC):	18-1D-EA-F4-37-B0
```

synicalsyntax, 2020-01-31:
Turns out I needed to initialize it through the official Anki app first, works fine now. Thanks

closed 2020-01-31 by synicalsyntax

faildproject, 2021-03-20:
I not understand the Workflow for connection.
The official SDK works fine, when i try your Lib i get the same error.
1.Cozmo is connected to Smartphone
2.Smartphone is connected via adp to Pc
3.SDK is enabled on Smartphone

Whats going wrong?