zayfod/pycozmo #21: Converting Audiokinetic WWave .wem files to PCM samples

zayfod, 2020-09-18:
The Cozmo app stores audio files in Audiokinetic WWave .wem format.

#20 Introduces support for playing back audio from files in 16-bit, mono, 22 kHz WAVE format.

A new module is needed that can read .wem files and convert them to PCM samples that can be played back through Cozmo directly.

The following repositories provide some base:
- https://github.com/hcs64/ww2ogg
- https://github.com/Crauzer/WEMSharp

FirefoxMetzger, 2020-12-03:
Do you happen to know where I can find the specification for the WEM format and could link it here? I recently wrote a format converter for DSDIFF to FLAC (delta-sigma modulated signal -> PCM) and this sounds somewhat similar. If it isn't too much hazzle I might be able to create a PR.

zayfod, 2021-01-04:
@FirefoxMetzger , unfortunately there is no pubic specification as this is a proprietary format by Audiokinetic.

It seems that internally the format uses different audio encodings. Cozmo and Vector .wem files seem to contain samples encoded primarily as OGG and ADPCM. The ww2ogg and WEMSharp linked above implement conversion of .wem files that use OGG encoding to regular .ogg files.

Having similar functionality in Pure python would be ideal for pycozmo.