zayfod/pycozmo #27: Firmware has reverted spontaneously to v10501 !

CozmoCyke, 2020-09-28:
%Run display_lines.py
2020-09-28 10:00:57.501 pycozmo.general      INFO     Firmware version 10501.
2020-09-28 10:00:57.501 pycozmo.general      WARNING  Factory/recovery firmware detected. Functionality is degraded.
2020-09-28 10:00:57.542 pycozmo.general      INFO     Body S/N 0x45a0330c, HW version 5, color 2.
2020-09-28 10:00:58.043 pycozmo.robot        ERROR    WiFi.Update: CLAD message 0x4b size 13 doesn't match size in buffer 9
2020-09-28 10:00:58.073 pycozmo.robot        ERROR    WiFi.Update: CLAD message 0x37 size 17 doesn't match size in buffer 18
2020-09-28 10:00:59.255 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.255 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.265 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.265 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.285 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.285 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.295 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.295 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.335 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.335 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.345 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.345 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.365 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.365 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.375 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.375 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.395 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.395 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.405 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.405 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.415 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.415 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.425 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.425 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.435 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.435 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.455 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.455 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.465 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.465 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.485 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.485 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.495 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.495 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.505 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.505 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.515 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.515 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.535 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.535 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.545 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.545 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.565 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.565 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.575 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.575 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:00:59.595 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
[... cut ...]
2020-09-28 10:01:21.025 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement
2020-09-28 10:01:21.025 pycozmo.general      ERROR    sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement

How to reproduce ?
I don't know but I don't launch at all tools to play with  safe file and write in the Cozmo memory like pycozmo_update.py. 

How to revert to last firmware ?
Can I use securely pycozmo_update.py or use the mobile app to update to last firmware ?

zayfod, 2020-09-28:
Sorry to hear that.

The Cozmo robot reverts to it's factory firmware any time it encounters a communication error (e.g. incorrectly formatted message) or issues in it's firmware.

While, `pycozmo_update.py` can be used to update to v2381, I'd recommend using the standard phone application for that as it is more robust. `pycozmo_update.py` is quite low-level, and requires you to extract the robot firmware (cozmo.safe) from a phone first.

The log above only shows Cozmo booting with the factory firmware, but does not reveal what happened before that.

Could you share the big WAVE file in some other way. Maybe Discord? The link above gives me HTTP error 403 - forbidden.

Do you see the "sendto() failed. [WinError 10035] Une opération non bloquante sur un socket n’a pas pu être achevée immédiatement" error with other examples also? It's a new error message, introduced in 0.7.0 and may reveal a need to improve the SendThread on Windows.

zayfod, 2020-09-30:
Based on the information in the description, I cannot identify a specific issue that can be worked on.

Closing for now.

closed 2020-09-30 by zayfod

legop3, 2023-04-03:
Hi, sorry to reopen this, but where do I get the firmware file from my phone? I want to try to implement a version of the update python script and make it automatic so that it updates each time before it tries to connect to the robot, because fixing the communication errors that cause it to factory reset is far outside of my scope. I am working on creating a web based remote controller, and I'm trying to automate the connection process. This will include flashing the update in case the robot got reset.