zayfod/pycozmo #34: Autonomy

benjamingranados, 2020-11-13:
Please make Cozmo autonomus 🙌🙌🙌🙌 The use of pycozmo in a autonomus way could make this little toy a truly robot pet for taking him to our jobs! Would love to see Cozmo in the desk without loosing connection!

zayfod, 2020-11-13:
@benjamingranados how do you think of Cozmo autonomy? What would you expect it to do on it's own? Could describe it in a bit more detail?

Also, what do you mean by "without loosing connection"?

benjamingranados, 2020-11-13:
Of course. What I meant is to be able to make Cozmo come back and get out of his charger by his own like in this autocozmo "cozmo_unleashed.py" script: https://gist.github.com/acidzebra/c02ff8c8ccb0e3a057ae0b48a5082a68

The problem with the preceding is that it connects and disconnects from the cellphone after some time when it is in charging state. I tried it with cozmo py, but of course I couldn't run it.

I would like Cozmo come and get out of his charger and when it's out it can just play as in its "exploration" game.

Thank you very very much for everything you're doing. I'll stay very tuned with your progress!

zayfod, 2020-11-13:
That's a good goal and I will for sure love to see such a script, based on PyCozmo also.

zayfod, 2020-11-15:
This seems to be a link to the same code in a GitHub repository by the same person - https://raw.githubusercontent.com/acidzebra/cozmo_unleashed/master/cozmo_unleashed.py

closed 2021-01-14 by zayfod