zayfod/pycozmo #37: RC car example with OpenCV UI

zayfod, 2020-11-15:
A new example is needed that allows controlling the robot as an RC car with OpenCV UI.

It should provide functionality, similar to the [rc.py]() example but also display a real-time image from the robot camera.

CozmoCyke, 2020-11-16:
Try this https://github.com/RecognitionDesigns/Vector_Object_Prediction_Recollection?fbclid=IwAR15nHwgUijuc6cCMhTV2Dg6T-6LsXLDvZMX592L8ldNteUGpF0bZlCLN-M

CozmoCyke, 2020-11-16:
Vector to Vector recognition : 
https://github.com/RecognitionDesigns/Vector-to-Vector-recognition

CozmoCyke, 2020-11-16:
https://github.com/RecognitionDesigns/Rock-Paper-Scissors-Vector-Game

zayfod, 2020-11-16:
The links are not directly related to the RC car example.

They are more appropriate for #35 or a more general approach for object recognition using CNNs.

The links don't seem very useful though. These are very basic experiments, the models, weights, or data sets do not seem to be shared.