zayfod/pycozmo #39: Cozmo as a Telepresence Robot?

chr0max, 2020-11-26:
Could this be possible?

Ok, I mean not like this 
[image]

At first, I love the idea Raspberry + Cozmo = ♥︎

I use a Raspi Zero with a Webcam and MotionEyeOS. I can connect through VPN at home from work and look what my cat is doing. Just for fun. I have the idea of using the good old cozmo to drive around and annoy or play with her. No serious cat-entertainment-plan, just for fun. But when you think about it, that could have other benefits. I mean, people are creative. I bet somebody would drive through his garden or do other crazy (legal- ☻) stuff with this remotely.
There is an older project controlling cozmo with a xbox 360 controller or a control-dashboard, but all with SDK and a mobile device. That premium solution is not really needed. A web-interface? sorry, I´m dreaming....

The first thing I was asking myself was "How to connect cozmo with my router?" 

What do you think?

FirefoxMetzger, 2020-12-03:
> The first thing I was asking myself was "How to connect cozmo with my router?"

You ... do not. At least not in the traditional sense. As far as I am aware, cozmo's wifi card is hard-coded to be an access point instead of a client. With enough hacking, you might be able to reverse engineer Anki's firmware and replace it with your own firmware that changes the Wifi settings, but that is outside of my area of expertise. 

What you can do is get a device (like a Raspi) with two interfaces that acts as a bridge; connecting one to cozmo (WiFi) and the other to your local network, and then pipe messages back and forth. You could envision running a small webserver (flask + pycozmo) that exposes an API to control cozmo or do it low-level by directly exposing a socket connection. That would get the remote control going.

For the webcam stream, you will face more trouble, because of latency. You likely want to re-encode the video into something suitable for streaming over the web and then publish that in a way that allows you to subscribe to a video stream. Unfortunately, nothing trivial.

Depending on where your remote is located you will have to figure out a smart way to deal with the latency between the camera stream arriving at the remote, and the control commands arriving at your cozmo and the resulting mismatch between the two to avoid crashing into things and making the controls feel somewhat responsive.

> The first thing I was asking myself was "How to connect cozmo with my router?"

I think it's doable (at least if the remote is in the same network), but it is certainly bigger than a quick weekend project / fun side tinker.

Good Luck!

chr0max, 2020-12-08:
Thanks for the Infos. 

Before I try to build a space rocket, I try to stay with a skateboard. So I think I could use the official app with a smartphone, that is connected through VPN to my home-network. So Cozmo has to be connected to the Raspberry and available in the network, so that the official app finds it. I will test it and hopefully the latency is enough for that.

closed 2021-01-14 by zayfod