zayfod/pycozmo #45: Eye/display problem

snowgoon88, 2021-01-24:
Hello everyone,

I think I have some problem with displaying things on my cozmo...

Preliminary : using the Anki android application, I have no problem with Cozmo display (display wifi info before logging, expressive eyes, games...).

When I run the pycozmo examples dealing with display:
- `display_image.py` : shows "Pycozmo" but sometimes also "strange" lines, like truncated diagonales (as if some values where also written in the display memory buffer).
- `display_lines.py` : works ok I think. I see a moving triangle

But, I have a black screen when I try:
- `procedural_face.py`
- `procedural_face_expressions.py`

(I have checked that `procedural_face_show.py` displays an image with two eyes using PIL on MY screen.

Besides, and I do not know if it is related, but I think that my first tries with pycozmo.client, the robot was displaying random eyes (a bit like the Andy app). Now, it keeps a black screen.

So I have questions (and maybe I'm using the pycozmo module the wrong way).

1) How can I debug the `procedural_face` and `procedural_face_expressions` scripts ? They should display something on the cozmo, no ?

2) What is the use of the `enable_procedural_face=False` while connecting to the client in these 2 scripts. I tend to think it is to stop the default display of random eyes. But in my case, with other application/examples, when it is set to `True` by default, nothing happens : black display for cozmo.

3) What am I doing wrong ? (Or what did I do wrong to make the default procedural face to be shut down "permanently" ?)

Snow/Alain
=========================
Some info on robot and software

Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux

pycozmo.__version__=0.8.0

When connecting to Cozmo:
021-01-24 19:12:45.050 pycozmo.protocol     INFO     Recv: 17B, 0F (disc.), 1F, 1P, 1P (100.00%); Sent: 0P, 0P (0.00%), 0F, 0F (disc.), 0B;
2021-01-24 19:12:45.074 pycozmo.general      INFO     Firmware version 2381.
2021-01-24 19:12:45.107 pycozmo.general      INFO     Body S/N 0x0ca05935, HW version 4, color 0.
2021-01-24 19:12:52.150 pycozmo.general      INFO     Done.

faildproject, 2021-03-21:
I have an other Problem with cli.display_image(im, 5.0), the duration time has no effect. Cozmo Displays the Image but falling back to eyes before duration time is finished.
I injected simply print("duration start" ) and print("duration stop") to the display_image function.
The sleep Part works but cozmo clears the Display before sleep finished.

gimait, 2021-04-20:
Hi @snowgoon88! 

I am not sure why you are seeing these issues. I have been trying to reproduce them now in a couple of cozmos and python versions, but things seem to be fine on my side.

From what you comment and looking around the code, I am thinking that the issue might be with the conversion from array (that comes from the procedural face generator) to PIL Images, or when loading the image in the `ImageEncoder`.

About your questions:
1. Debugging pycozmo and most of the scripts is not trivial, since stopping the scripts at any time makes the connection to the robot drop, which makes debugging a bit difficult. Nevertheless, you can easily try to debug it with any typical python debugging tool, if you clone the repository and run the scripts from there. The `procedural_face` and `procedural_face_expressions` scripts are quite similar: They work very much the same as the `display_image`, only using a generated image instead of one loaded from a file. This way, `procedural_face` loads a single image while `procedural_face_expressions` loads images of each expression and intermediate states are generated and sent to the robot. Both scripts should show cozmo's eyes with different expressions.
 
2. The `enable_procedural_face` flag is used to control wether the automatic generation of faces for cozmo is enabled. This is a feature more complex than what is shown in the example files, where you can load a procedural face statically. When `enable_procedural_face` is True, the robot will show its eyes moving as long as no other animation is shown. In the `procedural_face` and `procedural_face_expressions` scripts, this feature is disabled so that you can test different expressions without being confused by other movements of the eyes that might happen before or after the animation.

3. I wish I knew hahaha Now seriously, I believe something might be failing with PIL, but I'm not sure. How have you installed the package? what version of PIL and Pillow do you have installed?

-----

Hey @faildproject, I took a look at the issue you mentioned. I am working on a fix for it now, it is actually a quite small issue so we should have it ready soon.

ADebor, 2022-02-09:
Hi all !

Looks like I have issues similar to Snowgoon's when trying to display messages on Cozmo's screen. As a matter of fact, I think we may both have the exact same problem, as I also get strange diagonals, and can't see no eyes... and just like Snowgoon, `display_lines.py` works fine. 

I know this issue is opened for quite a long time now, but any help would be greatly appreciated. In particular, if @snowgoon88 solved the problem, please let me know !

Thank you in advance !

---

For all intents and purposes, here's what I get when running one of the examples:

2022-02-09 13:51:18.898 pycozmo.general      INFO     Firmware version 2381.
2022-02-09 13:51:18.944 pycozmo.general      INFO     Body S/N 0x43216893, HW version 5, color 3.
2022-02-09 13:51:36.374 pycozmo.general      INFO     Done.

and here are the packages and versions I'm using (with Python 3.7.0):

| Package |   Version |
| -----------  | -------- |
|dpkt     |   1.9.7.2|
|flatbuffers |2.0    |
|numpy      | 1.21.5| 
|Pillow    |  6.0.0  |
|pip       |  22.0.3 |
|pycozmo   |  0.8.0|  
|setuptools | 60.6.0|
|wheel    |   0.37.1|

bp88dev, 2025-06-16:
> I have an other Problem with cli.display_image(im, 5.0), the duration time has no effect. Cozmo Displays the Image but falling back to eyes before duration time is finished. I injected simply [...]

HOLY CRAP I'VE BEEN TRYING TO FIGURE THIS OUT BUT IT WAS JUST A BUG IN THE LIBRARY

bp88dev, 2025-06-16:
> HOLY CRAP I'VE BEEN TRYING TO FIGURE THIS OUT BUT IT WAS JUST A BUG IN THE LIBRARY

I figured it out: all you have to do is call cli.enable_procedural_face(False) before you do the image

zayfod, 2026-11-15:
Thanks everyone for the detailed reports, and sorry for the long silence!

@faildproject: the `display_image()` duration problem is fixed in #88. With procedural face enabled (the default), the displayed image was replaced by the face on the next frame, so duration had no visible effect. The fix is on master and will be in the next release. Until then, you can install it with `pip install --upgrade git+https://github.com/zayfod/pycozmo.git`.

@snowgoon88 @ADebor: the other problems (stray diagonal lines with display_image.py, black screen with `procedural_face.py` / `procedural_face_expressions.py`) I haven't been able to reproduce from the code alone. A lot has changed since 0.8.0 (Pillow, Python 3.11+ support, and several animation controller fixes), so could you check whether they still happen with the version from master? If they do, please post your Python, Pillow and pycozmo versions, and a photo of the screen if possible.

closed 2026-11-15 by zayfod