zayfod/pycozmo #54: Cozmo App doesn't connect to pycozmo server

reisxd, 2022-01-01:
Title says it all. And yes i modified Cozmos app to connect to my PC instead of cozmo (well... i don't even have a cozmo lol).

The engine sends "COZ\x03RE\x01" but the pycozmo server doesn't respond to it for some reason...?

reisxd, 2022-01-01:
This is probably a library problem. I tried connecting to the server on my PC from my phone and it didn't work and tried the opposite. Even installed pycozmo on my RPI Zero and connecting to it from my phone/pc and it still didn't work.

@zayfod Do you have any idea why?

reisxd, 2022-01-01:
Update again, modified Cozmos app once again to connect to the local server on my phone and it works (sort of, still doesn't go past "Connecting to Cozmo..." screen(I get the Connected message on the server though, might need to add missing packets)).

Why does this work locally?

reisxd, 2022-05-11:
Probably the last reply if this doesn't get any response from anyone else.
I today came back to this and I found out the Cozmo app repeatedly sending the Ping frame and I think PyCozmo sends something invalid, causing Cozmo app to send Ping frame again and again until it fails.

Also, the server I think doesn't support giving cube data because I got to the "Pull the tabs on your Cubes" screen (by forcing the app) and clicking continue after it appears in a few seconds (Note that the frame might not even got sent because the Cozmo app probably closes the TCP connection after it fails).

zayfod, 2023-12-24:
I am not sure I understand your setup.

PyCozmo is designed to interact with Cozmo and not the Cozmo mobile app.

closed 2023-12-24 by zayfod