zayfod/pycozmo #59: Still active?

Sojens, 2023-07-13:
Is this repository still active? Did zayfod go somewhere? Where do I go for new updates?

An issue I've encountered is that the speaker seems to cut out randomly, and animations seem to be a WIP (no audio, and very stuttery)

cozmo14047, 2023-12-10:
> Is this repository still active? Did zayfod go somewhere? Where do I go for new updates?
> 
> An issue I've encountered is that the speaker seems to cut out randomly, and animations seem to be a WIP (no audio, and very stuttery)

hi there i am thinking of bringing this back and maybe making a new app as i have seen that cozmo is going down i just need permision of the maker of this to allow me to make a new resporitory 
i will post a link here to mine could you please post a video of your problem

cozmo14047, 2023-12-11:
> Is this repository still active? Did zayfod go somewhere? Where do I go for new updates?
> 
> An issue I've encountered is that the speaker seems to cut out randomly, and animations seem to be a WIP (no audio, and very stuttery)

Here is my resporitory feel free to make requests or issues https://github.com/cozmo14047/cozmo-app

Sojens, 2023-12-19:
> Here is my resporitory feel free to make requests or issues https://github.com/cozmo14047/cozmo-app

I know I'm not a maintainer of this repo, but these types of self-advertising things are not the type of thing you post in GitHub issues, and what you posted is certainly not related to my original issue's topic. Please post this somewhere else.

zayfod, 2023-12-24:
The repo is not dead but I guess I am looking for inspiration and have not touched the code much lately.

closed 2023-12-24 by zayfod

icejix, 2023-12-26:
> The repo is not dead but I guess I am looking for inspiration and have not touched the code much lately.

Thanks for responding, even if you're not going to update the code much anymore. I'm still using this code almost daily, for a web-based project. Your code has a few minor issues, but it does work. Rodents are fun to play with.

Also, if you'd like to see what I've been doing sometime, I'd be happy to show you. Your code, even if it is buggy, has been a huge help, and without it, I don't know what I would've done to create this. Creating custom Rodents is a pretty difficult task, and I dont think I would have wanted to do that.

Good day to you!

Sojens, 2023-12-26:
> The repo is not dead but I guess I am looking for inspiration and have not touched the code much lately.

Thanks for responding, even if you're not going to update the code much anymore. I'm still using this code almost daily, for a web-based project. Your code has a few minor issues, but it does work. Rodents are fun to play with.

Also, if you'd like to see what I've been doing sometime, I'd be happy to show you. Your code, even if it is buggy, has been a huge help, and without it, I don't know what I would've done to create this. Creating custom Rodents is a pretty difficult task, and I dont think I would have wanted to do that.

Good day to you!