zayfod/pycozmo #67: No module named ´chunk´ and "from . import soundbank"

Breiwalker, 2025-04-16:
Idk what errors that are but Pls help me. Im also pretty new with Pycozmo!

RvBjx, 2025-05-31:
I encountered the same error and I am not sure if my (chatgpts) workaround is good but at least basic scripts run now. ChatGPTs answer: 
**Manually patch the library**

    Open the file:

C:\PROJECTDIRECTORY\.venv\Lib\site-packages\pycozmo\audiokinetic\soundbank.py
(Change PROJECTDIRECTORY for your scripts)

Find the line:

from chunk import Chunk

Replace it with:

from wave import _Chunk as Chunk

    ⚠️ _Chunk is a private API from the wave module, but it behaves similarly to the old chunk module. Use this only as a temporary workaround.

Save and rerun your script.

zayfod, 2026-11-15:
Thanks for reporting, and sorry for the slow reply!

This was caused by the `chunk` module being removed from the standard library in Python 3.13. It's fixed in pycozmo 0.9.0, which supports Python 3.11–3.15: `pip install --upgrade pycozmo`.

The `wave._Chunk` workaround is no longer needed.

closed 2026-11-15 by zayfod